vintagepc/MK404 #187: [FEATURE] Alternate visual options

vintagepc, 2020-07-27:
@gregsaun,expressed interest in alternate (:bear: ) visuals. Probably a feature we'll want to have in order to support other models of printer. 

Start thinking about ways we can do this, maybe a model table with offsets rather than reimplementing the entire MK3SGL class for every model. 

Maybe acquire a set of Bear visuals in order to facilitate this and see what problems come out of the woodwork during the process...

gregsaun, 2020-07-28:
Thank you for have noted my interested :)

I'll be happy to provide everything I can make to help you. You can already download the Bear 2.1 frame in Fusion 360 and STEP format from here: https://github.com/gregsaun/prusa_i3_bear_upgrade/tree/dev/src . The Bear extruder is available from here (same file formats): https://github.com/gregsaun/bear_extruder_and_x_axis/tree/master/source . Note the Bear frame repo does not contain the extruder because you can use the Prusa extruder on the Bear frame or Prusa frame with the Bear extruder.

leptun, 2020-07-28:
@vintagepc do you want me to make new OBJ files?

gregsaun, 2020-07-28:
If you use the STEP file to make OBJ files make sure to remove the "build helpers". There is one for Y axis and one for Z axis:
[image]
[image]

If building OBJ files is just an export from an STL (Meshmixer?) I can make those for you, no problem.

leptun, 2020-07-28:
For the time being I used Solidworks and a plugin to export as that is my main CAD software. I’m doing it like this just so that the output file is consistent.
Also, I had an assembly of the MK3 in Solidworks format and it was easier than using steps. Also, I could split the printer into subassemblies.

gregsaun, 2020-07-28:
Okay I see. Unfortunately I use Fusion 360 so STEP is probably best if you want to import to Solidworks.

However, I have just checked and Fusion 360 can export an OBJ. It takes a bit of time but I test it and share it here in case it looks to be good.

vintagepc, 2020-07-28:
Unfortunately we've been there before. Maybe you'll have better luck with a paid version if that's what you're using, but when I tried it with the maker license, the OBJs I got did not export materials, and you could not export multiple objects into one. 

FreeCAD does support MTL exports but the resulting appearance of the export was dissapointing given the number of triangles. There is a discussion on this in #50 with plenty of screenshots; in the end we concluded that Solidworks was the final export tool of choice. 

- You can use "GLC player" to examine the existing OBJs in the assets/ folder to see what I mean - each one contains multiple objects and materials, we've mostly broken them down by sub assemblies to swap them in/out as needed.

gregsaun, 2020-07-28:
I have the paid license but it looks like the material isn't exported.
[image]

vintagepc, 2020-07-28:
Darn - if you look at the 'Model' button on the left panel, are they all distinct components, or is it just one giant mesh? If the former, it may be possible to manually apply materials. But if it's all the same surface that's not possible. :disappointed:

gregsaun, 2020-07-28:
I have individual components yes. I cna share the OBJ and a prepared STEP file. The step file has the material (at least color).

vintagepc, 2020-07-28:
That's good - since it's just a text format I could just extract materials from the existing mtl files we have rather than manually migrate the data from the STEP. - the OBJ alone should be enough.

We'll have to see how it goes as a complete assembly, I think it's possible to rework things to handle that. But we might need to split things up into the separate moving components (X, Y, Z) for ease of handling within GL; depends how the object itself is structured. 

We had to split them for the other models because SW joins objects with the same material together - so if you tried to move the extruder, all other black parts would move left and right as well :rofl: (There was also lite mode which saved a lot of file loading overhead by doing that. But there's not much point in a Bear lite mode as it'd look about the same as the regular lite.)

gregsaun, 2020-07-28:
Looks like he has grouped some stuffs together so not sure how it goes. If you like I can create an OBJ for each axis + frame. If that could make things easier.

vintagepc, 2020-07-28:
Ah, OK. Lets do that then. 

Here are the assemblies we have for the existing model:
- Extruder, minus the two moving inner fan parts (already have separate objs for those to animate them)
- X axis (Z gantry)
- Y axis (minus steel sheet, already separate obj)
- Frame (minus LCD knob, separate obj already). 

In order to cut down on the triangle count and file size, you can also hide any internal objects that are not visible - for example the internal IR sensor ball, lever mechanism, some screws, the balls, if the LM8UUs have them modeled, etc)

I'm sure we'll find more things as we go along, this is usually an iterative process to get right.

gregsaun, 2020-07-28:
Awesome, I'll prepare that. What about belts (mine have no teeth)?

vintagepc, 2020-07-28:
We don't have any for the existing model but if it has no teeth it would look fine just leaving it on in the fixed parent assembly.

gregsaun, 2020-07-28:
Here it is :)
[bear21_mk3s_simulator.zip](https://github.com/vintagepc/MK404/files/4988499/bear21_mk3s_simulator.zip)

Thanks for all the hard work, very much appreciated.

vintagepc, 2020-07-28:
Thanks, I'll poke at it a bit after work tonight.

vintagepc, 2020-07-28:
This is interesting, the objs do have MTL references:

`mtllib d2b749dc-85a2-4cde-b3fc-98cb260eb702.mtl`

...
`usemtl Aluminum_-_Satin`

Do you perhaps need to download or retrieve them separately? I can't imagine they'd go through all the trouble of handling it and not exposing the materials file.

3d-gussner, 2020-07-29:
I may have found a way to "export" Fusion 360 to obj + mtl.
1. Disable history
2. Convert Body "BRep to Mesh"
3. Save as OBJ

Result:
<name>.obj + <name>.mtl
[Cubes.zip](https://github.com/vintagepc/MK404/files/4993175/Cubes.zip)

Now have to try an assembly.

gregsaun, 2020-07-29:
@vintagepc Very happy you like the Bear. Contact me first if you like to build a Bear from scratch ;)

@vintagepc @3d-gussner Oh that is nice, let me redo the export then!

3d-gussner, 2020-07-29:
https://apps.autodesk.com/FUSION/en/Detail/Index?id=161925079138285899&appLang=en&os=Win64 Seams to work. Here an example of a Zaribo:
[image]
[image]

gregsaun, 2020-07-29:
Here are the OBJ, I used the proposed @3d-gussner plugin, it 
[bear21_mk3s_simulator_02.zip](https://github.com/vintagepc/MK404/files/4994669/bear21_mk3s_simulator_02.zip)
seems to work well.

I made a complete assembly OBJ and also separated axis OBJs.

3d-gussner, 2020-07-29:
> I made a complete assembly OBJ and also separated axis OBJs.

[image]
Looks like some components don't have the correct material ;-)

gregsaun, 2020-07-29:
Yeah I save some time with not setting all the material. I have a print settings and BOM for that ;)

vintagepc, 2020-07-29:
Let's not get too caught up - the materials are trivial to manipulate, so it's relatively easy to swap one for the other. I can work with what we have, I think, so the time spent trying to get it to export perfectly is probably more than the time for me to just edit the file.  :+1: 

I'm working on the model-swapping framework now, once that's in hopefully it's pretty trivial to add the models and take it from there.

gregsaun, 2020-07-29:
If I could avoid to change all materials and redoing a exports, pushing to github source files and so on I would appreciate :D . But if this become a too big work I can figure it out. Why is the type of material important? I mean what is important is to have different material, isn't it?

vintagepc, 2020-07-29:
The mtl files that are paired with the OBJ define how the object renders - that is why the material as a whole matters, not just the name. If you look in the .mtl file with a text editor, then you can see that each material has a set of properties that define things like colour, reflectiveness, etc. and these are used directly to set the appearance of the polygons during the render.

It would have been very nice to be able to export the obj+mtl together and use as-is without extra work, but as it is we'd either need to spend a bunch of time fiddling with the source CAD files to get it to export correctly via some ill-defined pathway, or I can spend an hour or so changing material names in the .obj/mtl file by hand. That only gets unpleasant if models have to be updated regularly.

Results are viewable with GLC_player. so if you'd like to try your hand at it, it is relatively straightforward, if a little tedious:
- Open the .obj file with a good text editor that can handle large files. 
- If changing the .mtl file, edit the `mtllib` line to match the name of the new .mtl file. May or may not be needed depending on what we got from the export, I haven't looked at the new files yet. Materials can be transplanted from one .mtl file to another by copy-pasting their text blocks. 
- Find the `usemtl` command for the body you wish to change, and edit the material name to the new one. 
- Reload the obj in glc_player and observe the changes.
- Repeat. the previous two steps until satisfied. :grin:

gregsaun, 2020-07-29:
Okay that make sense, in this case I prefer to set properly everything in CAD (I don't like to redo stuffs). However, my parameters might be different than the other Prusa you already have in your simulator (but probably closer).

In Fusion 360 there is the appearance and the physical material. Looks like MTL take only the appearance into account, am I correct?

vintagepc, 2020-07-29:
Perhaps that is it - unfortunately I cannot say for sure as I haven't dived into the MTL exports from Fusion after our initial failure with the 3S model. We found it just worked from Solidworks so went with that, also since it was the format of the original source we used. 

Should be easy to identify though with a simple test model that has two cubes, one with a material and one with an appearance set, and seeing which one comes out correctly.

It doesn't help that the term "material" is being used in different contexts in the different tools :roll_eyes:

gregsaun, 2020-07-29:
Okay after more testing it does not seem to have trace of physical material from Fusion 360, it only uses appearance. That makes my job a little easier then. Do you still want separated OBJ for all axis or doing a single one is enough for you now that I know how to set things correctly?

vintagepc, 2020-07-29:
That's excellent!

I would still prefer separate objects for each axis (essentially, collection of bodies that need to move together, independent of other bodies). It's probably possible to do it with a single OBJ, but I don't think that's very maintainable longer term - it would fall apart pretty quickly if the models changed.

vintagepc, 2020-07-30:
Lighting weirdness aside, making progress!

[image]

gregsaun, 2020-07-30:
Hu that is some progress! I should share the OBJ with proper material soon ;)

vintagepc, 2020-07-30:
I may end up editing the materials slightly at the end, it looks like Fusion only exports a Kd (diffuse) parameter. 
This results in a very muted appearance in the object. For example, the knob and SD card (which also have ambient and specular parameters) vs the display housing:

I'll have to dive into the lighting again to see if that's truly the case since it doesn't render as poorly in GLC player. We'll see. 

[image]

gregsaun, 2020-07-30:
Okay let me know if I can help you in a way. I used the plugin maybe I can try again from Fusion 360 directly if you like.

Here is the export: 
[bear21_mk3s_simulator_03.zip](https://github.com/vintagepc/MK404/files/5002396/bear21_mk3s_simulator_03.zip)

I have included a full export in case you need some other parts like the know and fans. Note that the hotend fan in the full assembly has been designed by myself and looks like a Sunon. The LCD knob is a little different but very close. The print fan is also my own model and should be close.

gregsaun, 2020-07-30:
Wait, it looks like the extrusion does not have the correct rendering. Let me export it again.

vintagepc, 2020-07-30:
Ah, OK - if you are using different fan models/knobs, then feel free to export those models separately too. The outer "shell" of the fan can remain part of the E assembly, I would just need the rotor as a standalone piece because we spin it if the fan is on.

gregsaun, 2020-07-30:
Okay I will do that!

gregsaun, 2020-07-30:
Okay here you have everything:
[bear21_mk3s_simulator_04.zip](https://github.com/vintagepc/MK404/files/5002564/bear21_mk3s_simulator_04.zip)

I have included two version of the extruder, with and without the hotend fan "outer shell". If it is too much work for you go with the fan and knob you already use.

vintagepc, 2020-07-30:
Great. The added work to swap out the model is minimal, the bulk of it just comes from setting up the positioning relative to the model - and that's also fairly minor. 

The real bear (heh) is the backend framework, but that's not really an impediment, we need it anyway since the intent is that we'll ultimately support completely different printer "models" which may not even look anything like a MK3, so a layer of abstraction between the drawing and the actual model components is a good thing to have. :+1:

gregsaun, 2020-07-30:
That all sounds very nice, looking forward to test this :)

vintagepc, 2020-07-30:
Whoops- looks like you left the print fan shroud hidden by accident in the E export.

gregsaun, 2020-07-31:
This is normal, the shroud is completely different than the Prusa one.

vintagepc, 2020-07-31:
Sorry, maybe I'm misunderstanding - I'm referring to the outer housing of the part cooling fan, the 5150 blower. 

The hot-end cooling fan shroud is present and accounted for :+1:

Getting close, I now just need to fix the lighting and un-break all the things I broke in the "standard" 3d models 😁

gregsaun, 2020-07-31:
Ha sorry, you can reuse the complete print fan you have for Original Prusa for that :). Looking forward to see it!

vintagepc, 2020-08-01:
Getting closer, I think I've sorted the lighting. 

[image]

The motors looking weird is a WIP, I've filed a bug/info request with https://github.com/tinyobjloader/tinyobjloader/issues/279

Part fan shroud is still missing, I need to extract the model from the 3S extruder first. 

Hopefully I can wrap this up tomorrow!

gregsaun, 2020-08-01:
It start to look good! I was wondering how the motors would render due to the material set on faces...

Do you want the print fan shroud without the blades? Mine is one piece normally and has no blade but I can try to remove them for you if you like.

Quick question, I saw you are using a different color for the LCD screen, is there a reason for that? I am asking because I have made an Inkscape SVG project for writing the menu in my assembly guides so it would be nice to use the simulator for that. Just a nice to have, nothing urgent or mandatory ;)

vintagepc, 2020-08-01:
That particular color combo is easier to read in the 3d graphics. the 2D display/control panel view is blue and white by default:

[image]

Also, I'm partial because that's what my real display looks like... :grin: 

[image]

gregsaun, 2020-08-01:
Ok I see, hard to read I agree. I was suspecting you did upgrade the real LCD screen haha :D

vintagepc, 2020-08-01:
If you can easily export the shroud sans the center blade, that would save me some time - thanks.

gregsaun, 2020-08-01:
I'll do it now

vintagepc, 2020-08-01:
I posted a guide on the display mod here:
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles-.../a-very-much-needed-improvement/

It's not an off-the shelf part or LCD module.

gregsaun, 2020-08-01:
Oh that was you, nice!

gregsaun, 2020-08-01:
I know I have seen your nickname few times :D

Here is the print fan: 
[bear21_mk3s_simulator_print_fan.zip](https://github.com/vintagepc/MK404/files/5010518/bear21_mk3s_simulator_print_fan.zip)

vintagepc, 2020-08-01:
It is dangerous to be a man of many skills and with far too much free time... :rofl: 

Thanks, will get that shroud in at some point.

gregsaun, 2020-08-01:
Oh yeah I know very well what you mean hahaha

vintagepc, 2020-08-01:
I think I've got all the issues ironed out, though I turned the per-face material issue into a separate bug.
`./MK404 -g bear`

closed 2020-08-02 by vintagepc

gregsaun, 2020-08-02:
Thank you very much!

I tried to launch your latest release and I get this error:
```
$ ./MK404 -h
./MK404: error while loading shared libraries: libGLEW.so.2.0: cannot open shared object file: No such file or directory
```

I have compiled master branch and I can launch it. To compile it successfully I had to add few libraries including the one linked in the error message. However, I have a "printer not found" message on the LCD emulator window.

For information, I am on a VM through virtualbox with fresh Ubuntu installation (minimal).

vintagepc, 2020-08-02:
hmm... what glew version does the system have? I'm using 2.1.0

gregsaun, 2020-08-02:
I have 2.1.0 as well. The symbolic links worked: `ln -s /usr/lib/x86_64-linux-gnu/libGLEW.so.2.1.0 /usr/lib/libGLEW.so.2.0` .

vintagepc, 2020-08-02:
awesome! :tada:

[2 more replies]

PR #191 (2020-08-02): 187 bear visual model
### Description

- Adds capability to change the printer 3d models by abstracting the objs from the primary drawing code. 
- Adds bear model files and model. (No MMU support)

### Behaviour/ Breaking changes

- adds a -g bear option. 
- Possibly broke the standard models somehow that I missed during testing.

### Have you tested the changes?

Tested locally with all three variants (lite, bear, fancy). All components appear to be in place and rendering properly. 

### Other

- Uncovered a bug, our TinyObjLoader code does not handle objects with per-face materials. (#190)
- [... cut]