vintagepc/MK404 #188: [BUG] X Error of failed request:  GLXBadContextTag

sl1pkn07, 2020-07-27:
**Describe the bug**
A clear and concise description of what the bug is. Please make sure you've searched for existing bug reports before creating a new one.

run fail when launch MK404

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Build project with:

cmake 3.18.0
gcc 10.1.0
glew 2.2.0
glu 9.0.1
sdl 1.2.15 (i'm not sure if use this one or sdl2)
sdl2 2.0.12 (i'm not sure if use this one or sdl1)
2060RTX (Nvidia vulkan drivers 440.66.17)
KDE 5.19.3
picocom 3.1

**Expected behavior**
run

**Screenshots**

~~~
┌─┤[$]|[sl1pkn07]|[sL1pKn07]|[/tmp/makepkg/mk404-git/src/MK404/build]|
└───╼  ./MK404 
ScriptHost: NOTE: Duplicate context name (Thermistor) with different pointer. Incrementing ID...
ScriptHost: NOTE: Duplicate context name (Thermistor) with different pointer. Incrementing ID...
ScriptHost: NOTE: Duplicate context name (Thermistor) with different pointer. Incrementing ID...
ScriptHost: NOTE: Duplicate context name (Fan) with different pointer. Incrementing ID...
Creating pinspec for atmega2560
Loading 262144 bytes of XFLASH
Read 262145 bytes
Loading 4096 bytes of EEPROM
Read 4096 bytes
NOTE: No firmware load requested, Einsy executing purely from flash memory.
read_ihex_chunks: stk500boot_v2_mega2560.hex, unsupported check type 03
Loaded 4202 bytes from HEX file: stk500boot_v2_mega2560.hex
Initialized VCD file Einsy_atmega2560_VCD.vcd
uart_pty_init bridge on port *** /dev/pts/4 ***
Connect: /tmp/simavr-uart2 now points to /dev/pts/4
note: export SIMAVR_UART_XTERM=1 and install picocom to get a terminal
uart_pty_init bridge on port *** /dev/pts/5 ***
Autodetected SD image size as 256 Mb
Init on ADC 0 start 25.00
Init on ADC 2 start 25.00
Init on ADC 3 start 25.00
Init on ADC 6 start 25.00
LCD: 37uS is 592 cycles for your AVR
LCD: 1uS is 16 cycles for your AVR
MK3S - adding IR sensor.
Validating script...
Script validation finished.
Starting atmega2560 execution...
MCUSR: 01
RESET
Button released after 16 usec
MCUSR: 00
OnCmdReady activating 4 bits mode
X Error of failed request:  GLXBadContextTag
  Major opcode of failed request:  151 (GLX)
  Minor opcode of failed request:  26 (X_GLXMakeContextCurrent)
  Serial number of failed request:  623
  Current serial number in output stream:  623
~~~

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

Archlinux

**Additional context**
Add any other context about the problem here - printer model, what firmware file you are running, etc.

build with:

~~~
  cmake -S MK404 -B MK404/build \
    -DCMAKE_BUILD_TYPE=None \
    -DCMAKE_INSTALL_PREFIX=/usr

  make -C MK404/build VERBOSE=1 Einsy_atmega2560_SDcard.bin UpdateDocs
~~~

the gcode file in SDCard is a simply 3DBenchi gcoded by prusa-slicer

vintagepc, 2020-07-27:
Hm, this looks like an issue with NV-GLX.  I get weirdness using primus/optirun on my hybrid machine too, but mesa is fine.

sl1pkn07, 2020-07-27:
same with 450.56.02

vintagepc, 2020-07-27:
🤔 Unfortunately I don't get GLXBadContext, but I get a black window for the LCD section and a weirdly scaled one for the 3d visuals. But the underlying cause is probably the same. I'll dig in tonight.

vintagepc, 2020-07-27:
I'm still not sure what's going on here. I'm by no means a GL person, so my best guess is something about the nvidia GL extension or drivers is unhappy. Unfortunately that means figuring this out is going to likely take a good chunk of time

I can't reproduce your exact problem even on a machine with a dedicated 1050 (430) the closest I get is black/blank windows under the nvidia driver. I'm hopeful if I resolve one it will resolve the other.

Does it start working correctly if you temporarily change to nouveau as your video driver?

sl1pkn07, 2020-07-27:
nouveau is no-op for me. i preefer wait

greetings

vintagepc, 2020-07-28:
@leptun @3d-gussner @wavexx could you chime in with what video drivers are on the system you've been using with MK404? Curious to see what data points we have and whether anyone has tried ATI/fglrx or a working nvidia setup.

3d-gussner, 2020-07-28:
I run hyper-v Ubuntu VM not sure what gpu is used. Have a look tomorrow

vintagepc, 2020-07-28:
OK, most likely Mesa/software GL then.

wavexx, 2020-07-28:
I'm using linux on the iHD Kabylake Intel drivers (kms). This is mesa with dri3 acceleration.

But truth be told, I remember early on I also had similar issues when starting without the fancy graphics triggering a bad context too. There's some timing issue IMHO.

vintagepc, 2020-07-28:
OK, sounds like you're probably also using the Mesa stack. 

Yeah, in the early stages there was definitely a bad init/setup that got fixed after I did a GL cleanup merge. 

But you have a good suggestion that may offer some insight - @sl1pkn07 what happens if you launch with `MK404 -f <firmware file> -g fancy`? Does it still choke?

sl1pkn07, 2020-07-28:
do more things, but finaly crash with same error

~~~
└───╼  ./MK404 -f MK3S.afx -g fancy
ScriptHost: NOTE: Duplicate context name (Thermistor) with different pointer. Incrementing ID...
ScriptHost: NOTE: Duplicate context name (Thermistor) with different pointer. Incrementing ID...
ScriptHost: NOTE: Duplicate context name (Thermistor) with different pointer. Incrementing ID...
ScriptHost: NOTE: Duplicate context name (Fan) with different pointer. Incrementing ID...
Creating pinspec for atmega2560
Loading 262144 bytes of XFLASH
Read 262145 bytes
Loading 4096 bytes of EEPROM
Read 4096 bytes
Loaded 227176 .text at address 0x0
Loaded 1120 .data
Loaded 228296 bytes from ELF file: MK3S.afx
read_ihex_chunks: stk500boot_v2_mega2560.hex, unsupported check type 03
Loaded 4202 bytes from HEX file: stk500boot_v2_mega2560.hex
Initialized VCD file Einsy_atmega2560_VCD.vcd
uart_pty_init bridge on port *** /dev/pts/3 ***
Connect: /tmp/simavr-uart2 now points to /dev/pts/3
note: export SIMAVR_UART_XTERM=1 and install picocom to get a terminal
uart_pty_init bridge on port *** /dev/pts/4 ***
Autodetected SD image size as 256 Mb
Init on ADC 0 start 25.00
Init on ADC 2 start 25.00
Init on ADC 3 start 25.00
Init on ADC 6 start 25.00
LCD: 37uS is 592 cycles for your AVR
LCD: 1uS is 16 cycles for your AVR
MK3S - adding IR sensor.
# of vertices  = 19840
# of normals   = 19840
# of texcoords = 19840
# of materials = 7
# of shapes    = 7
shape[0] # of triangles = 3780
shape[1] # of triangles = 3456
shape[2] # of triangles = 1158
shape[3] # of triangles = 4110
shape[4] # of triangles = 3654
shape[5] # of triangles = 2724
shape[6] # of triangles = 1446
m_extMin = 0.023854, 0.043530, 0.155235
m_extMax = 0.277659, 0.087820, 0.436635
# of vertices  = 5188
# of normals   = 10226
# of texcoords = 0
# of materials = 3
# of shapes    = 3
shape[0] # of triangles = 9212
shape[1] # of triangles = 82
shape[2] # of triangles = 932
m_extMin = 65.015419, 26.029099, 482.844208
m_extMax = 97.015419, 43.888569, 500.724396
# of vertices  = 6163
# of normals   = 6163
# of texcoords = 6163
# of materials = 2
# of shapes    = 2
shape[0] # of triangles = 1724
shape[1] # of triangles = 3208
m_extMin = 0.000146, 0.000000, -0.241146
m_extMax = 0.252145, 0.001010, -0.000145
# of vertices  = 1047
# of normals   = 1047
# of texcoords = 1047
# of materials = 1
# of shapes    = 1
shape[0] # of triangles = 1112
[... 137 lines cut ...]
shape[51] # of triangles = 5212
shape[52] # of triangles = 54
shape[53] # of triangles = 6490
m_extMin = -0.037741, 0.015460, 0.102499
m_extMax = 0.345554, 0.392379, 0.524248
# of vertices  = 2280
# of normals   = 4588
# of texcoords = 0
# of materials = 2
# of shapes    = 1
shape[0] # of triangles = 4588
m_extMin = 46.971825, 359.662842, 261.699463
m_extMax = 85.971825, 397.184845, 289.699463
Validating script...
Script validation finished.
Starting atmega2560 execution...
MCUSR: 01
RESET
Button released after 16 usec
X Error of failed request:  GLXBadContextTag
  Major opcode of failed request:  151 (GLX)
  Minor opcode of failed request:  26 (X_GLXMakeContextCurrent)
  Serial number of failed request:  729
  Current serial number in output stream:  729
~~~

vintagepc, 2020-08-02:
For my own reference, MVE with the same issue. Works with MESA, black screen with nvidia.
(rename to .cpp)
[example.txt](https://github.com/vintagepc/MK404/files/5012432/example.txt)

vintagepc, 2020-08-02:
I think I may have found the cause. Apparently freeglut does not like being pushed to a background thread, and doing so causes all kinds of interesting issues.

@sl1pkn07  can you check out and try this branch?
https://github.com/vintagepc/MK404/tree/188-freeglut-threading

if it fixes the issue for you I'll clean it up and merge.

sl1pkn07, 2020-08-02:
[image]

weee

sl1pkn07, 2020-08-02:
also, i have some changes in the CMake Scripts for use SDL2 instead SDL(1) and chech GL, GLUT and GLEW througth CMake
(also use speces instead of tabs)

~~~patch
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 15e6929..28c2897 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -22,115 +24,115 @@ set (FIRMWARE
     )

 set(H_FILES
-	parts/ADCPeripheral.h
-	parts/PinNames.h
-	parts/BasePeripheral.h
-	parts/boards/EinsyRambo.h
-	parts/boards/MM_Control_01.h
-	parts/PrinterFactory.h
-	parts/wiring/Einsy_1_1a.h
-	parts/wiring/MM_Control_01.h
-	parts/wiring/Einsy_1_0a.h
-	parts/ScriptHost.h
-	parts/pinspecs/PinSpec_2560.h
-	parts/pinspecs/PinSpec_32u4.h
-	parts/PinSpec.h
-	parts/SoftPWMable.h
-	parts/TelemetryHost.h
-	parts/Printer.h
-	parts/SPIPeripheral.h
-	parts/Scriptable.h
-	parts/Board.h
-	parts/components/LED.h
-	parts/components/PINDA.h
-	parts/components/HD44780GL.h
-	parts/components/SDCard.h
-	parts/components/uart_pty.h
-	parts/components/hd44780_charROM.h
-	parts/components/UART_Logger.h
-	parts/components/RotaryEncoder.h
-	parts/components/EEPROM.h
-	parts/components/Thermistor.h
-	parts/components/Fan.h
-	parts/components/ADC_Buttons.h
-	parts/components/VoltageSrc.h
-	parts/components/w25x20cl.h
-	parts/components/Beeper.h
-	parts/components/HD44780.h
-	parts/components/SerialLineMonitor.h
-	parts/components/HC595.h
-	parts/components/Button.h
-	parts/components/MMU2.h
-	parts/components/PAT9125.h
-	parts/components/GCodeSniffer.h
-	parts/components/TMC2130.h
-	parts/components/IRSensor.h
-	parts/components/Heater.h
-	parts/IScriptable.h
-	parts/PinSpec_Helper.h
-	parts/Wiring.h
-#	parts/IRQHelper.h  Can't be checked/used standalone.
-	parts/printers/Prusa_MK3S.h
-	parts/printers/Prusa_MK3SMMU2.h
-	parts/printers/Prusa_MK3MMU2.h
-	utility/Color.h
-	utility/GLPrint.h
-	utility/FatImage.h
-	utility/MK3SGL.h
-	utility/GLObj.h
-	utility/OBJCollection.h
-	utility/SerialPipe.h
[... 295 lines cut ...]
+#include <SDL2/SDL.h>         // for SDL_Init, SDL_INIT_AUDIO
+#include <SDL2/SDL_audio.h>   // for SDL_PauseAudio, SDL_AudioSpec, SDL_Clos...
+#include <SDL2/SDL_error.h>   // for SDL_GetError
+#include <SDL2/SDL_stdinc.h>  // for Sint16
 #include <stdio.h>            // for fprintf, printf, stderr
 #include "BasePeripheral.h"   // for MAKE_C_CALLBACK
 #include "TelemetryHost.h"
diff --git a/parts/components/Beeper.h b/parts/components/Beeper.h
index 3c166fb..74c3cdc 100644
--- a/parts/components/Beeper.h
+++ b/parts/components/Beeper.h
@@ -21,7 +21,7 @@

 #pragma once

-#include <SDL/SDL_audio.h>  // for SDL_AudioSpec
+#include <SDL2/SDL_audio.h> // for SDL_AudioSpec
 #include <stdint.h>         // for uint16_t, uint8_t, uint32_t
 #include <string>           // for string
 #include <vector>           // for vector
~~~

is ok for you?, i can send a PR

greetings

vintagepc, 2020-08-02:
You would be welcome to submit a PR, but please do not convert tabs to spaces. Preference here is tabs, because I'm firmly in the "tabs can be customized to suit a user's indentation preferences" camp and not the "it should look the same for everyone" camp :wink:

PR #197 (2020-08-02): 188 freeglut threading
### Description

Moves glut back to the main thread.

### Behaviour/ Breaking changes

Changed quit handling, may have improved #103, may have broken something else.

### Have you tested the changes?

Yes. both GL windows now render correctly with Mesa and nvidia GL.

### Linked issues:

- fixes #188
closed 2020-08-02 by vintagepc