vintagepc/MK404 #201: [BUG] Vanilla Marlin for Einsy Rambo not working in MK404 sim.

3d-gussner, 2020-08-04:
**Describe the bug**
Got vanilla Marlin for Einsy Rambo kind of working and it can be flashed on a physical printer.
Trying to load the same `hex` file in MK404 doesn't do anything.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile https://github.com/3d-gussner/Marlin/tree/MK3s_2.0.x
2. Copy `\.pio\build\rambo\firmware.hex` to your MK404 build folder
3. Start MK404 `/build/MK404 -f firmware.hex
4. MK404 starts but no output on LCD on terminal

**Expected behavior**
It would behave like the real Einsy board.

**Desktop (please complete the following information):**
- Hyper-V VM
 - OS: [Ubuntu]
 - Version [18.04.1]

[image]

vintagepc, 2020-08-04:
Have you tried running with `-n` (aka `--no-hacks`)? This is necessary because the TIMSK tweak we need to get the firmware to boot has the opposite effect on stock Marlin.

3d-gussner, 2020-08-04:
Sorry didn't find this issue #117

Solved

closed 2020-08-04 by 3d-gussner

vintagepc, 2020-08-04:
Not a problem :)

Maybe when I figure out #75 this workaround can go away and both work without it

vintagepc, 2020-08-05:
Good news (also @leptun ) - `--no-hacks` is no longer necessary for Marlin thanks to closing #75.