vintagepc/MK404 #205: [Q+A] Which library provides uart_pty_fifo_write?

bozzzzo, 2020-08-07:
Trying to compile on OSX...

I'm blocked at compiling `uart_pty.cpp` due to undeclared identifiers `uart_pty_fifo_{read|write|isempty|isfull}`.

Where are these suppoed to come from? Google also doesn't know much about them...

bozzzzo, 2020-08-07:
Ah, they should come from those macros...

wavexx, 2020-08-07:
On Fri, Aug 07 2020, Bozo Dragojevic wrote:
> Trying to compile on OSX...
>
> I'm blocked at compiling `uart_pty.cpp` due to undeclared identifiers
> `uart_pty_fifo_{read|write|isempty|isfull}`.

These are part of the source inside uart_pty itself.
Showing the output of cmake and the full build log would help.

vintagepc, 2020-08-07:
Correct, they come from `fifo_declare.h` inside SimAVR, which has the `DEFINE_FIFO` and `DECLARE_FIFO` macros.

vintagepc, 2020-08-07:
BTW - if you do get a successful OSX build without too many breaking or divergent changes, I'd definitely be interested in adding a CI build/workflow for OSX release binaries. I have no means to continue maintaining or testing it, but I'm hoping things are not too dissimilar between the two systems that it's impossible.

bozzzzo, 2020-08-07:
So, I figured this out, expect a PR soon

closed 2020-08-07 by bozzzzo

vintagepc, 2020-08-07:
Nice work!