vintagepc/MK404 #218: [BUG] Broken graphics

leptun, 2020-08-10:
**Describe the bug**
The graphics seem to be broken in the latest master branch. I'm using Prusa_MK3SMMU2 fancy graphics.
Stuff that I can see is broken:
- The bed LED is floating mid-air, but it's working
- One of the MMU motors seems to blink with green graphics on startup sometimes
- The extruder visualizer is turning much too little.

**To Reproduce**
Steps to reproduce the behavior:
`./MK404 -s Prusa_MK3SMMU2 -g fancy -m`

**Expected behavior**
It should work :)

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
Ubuntu 18.04.4 LTS - VMware

vintagepc, 2020-08-10:
whoops! probably some side effects from the Bear visuals rework.

leptun, 2020-08-10:
Also, just noticed this as well: The backlight is stuck on 100% brightness

vintagepc, 2020-08-10:
oh, good. I thought I broke that during my cleanup in #217 ... was scratching my head why.

vintagepc, 2020-08-10:
I didn't think the MK3 OBJ files had any per-face colours, but what probably happened is #190 changed the indexing for the objects we show and hide.

vintagepc, 2020-08-12:
Just noticed we're also no longer showing the correct E variant (MMU/std)

vintagepc, 2020-08-12:
Looks like the brightness was broken by https://github.com/vintagepc/simavr/commit/e838af1472e8402f4260e6432cbbe6d6a6fe6a85

Investigating...

PR #221 (2020-08-13): 218 broken visuals
### Description

Fixes a bug and some busted visuals as a result of the bear cleanup

### Behaviour/ Breaking changes

Bug in GLObj would cause items to have inflated object indices, meaning any calls to set an object's properties by index would be changing the wrong object.

LCD brightness is still broken (will fix before merging)

### Have you tested the changes?

Yes - MMU LEDs appear correct now, heatbed LED is in the correct place for both sets of visuals, sheet hides properly with `T` key, E-vis now spins a bit faster.

### Linked issues:

 - Closes #218
closed 2020-08-13 by vintagepc