vintagepc/MK404 #219: [Q+A] How to add new printer with new board?

3d-gussner, 2020-08-11:
I would like to add the Prusa MK2, MK2.5 and MK2.5S to MK404.
Here is what I think i have to do:
- add miniRAMBo10a to parts/wiring
- add miniRAMBo13a with overwrites to parts/wiring
- add miniRAMBO to parts/boards
- add A4982 to parts parts/components
- add Prusa MK2, MK2.5 and MK2.5s to parts/printers
- add MK2 frame as obj
- add MK2 extruder as obj
- add MK2.5/s extruder as obj
- add other MK2/MK2.5/s objs
- add MK2... to CMakeLists.txt
- add MK2... to MK404.cpp
- add MK2... to PrinterFactory.h
- add MK2... to launch.json
- add MK2... to CommandLine.md

I am not sure how to get the A4982 working correctly, the rest I can try.

Do we need extra endstop switches in components? Very similar to IR except analog voltage reading.

How would you begin with this to create a PR?

Also saw that the parts/wiring/Einsy_1_?a.h files don't use the schematics pin names instead the connected component or in printer defined names.
Example:
Einsy schematics PJ2 on P2 pin8 is called BTN_EN1 pin 72
For the Prusa MK3/s makes sense, but what if somebody decides to use an I2C display or something else on EXT2/P2 connector?

vintagepc, 2020-08-11:
I think you have  most of the outline right (no need to update the various .md files - those are autogenerated. You only need to add entries to the document generation section to have it create any printer-specific telemetry or scripting documentation.

- You should not need to add anything to MK404.cpp, printer type is all abstracted away into other classes.
- Endstops: I think you could probably even get away with re-using either ADCButtons or the standard Button component. 
- Wiring is an abstraction layer between the AVR pins and the board's outputs/convenience names. It's arbitrary, you could easily add a new (`MY_LCD_MISO`) pinname entry  and map it to a P2 pin for use in a custom board. 

Since you're a collaborator you should have write access to the repo - feel free to create a branch and push it up so we can collaborate easily and incrementally. 

Let's walk through it step-by-step in the following order. I am happy to contribute or review at each step. (I'm hoping to get #216 wrapped soon, but it's a lot of work... ;))
1.  Add pinspec for the AVR, if necessary.  (Likely not needed, miniRambo is a 2560)
2. Define a wiring with accessory names to MCU pin numbers.
3. Define a board with the given wiring/pinspec. 
4. Add hardware to the board. 
5. Create the printer object.
6. Add to PrinterFactory
7. 3D assets and visual work. (Defining an Obj collection with the appropriate OBJs, wiring up to the printer).  

Does that help clear up the process?

3d-gussner, 2020-08-15:
Created new branch https://github.com/vintagepc/MK404/tree/miniRAMBo_MK2_MK25_S

vintagepc, 2020-08-15:
I had a peek at the A4982 datasheet, seems pretty straightforward to implement. Would you like me to do that? I can start templating out the board and printer too.

3d-gussner, 2020-08-15:
It would be great if you could do the stepper driver.

vintagepc, 2020-08-15:
Sure thing!
I pushed up a barebones 2.5 1.3 MiniRambo printer, it boots but of course there's not much else in the way of hardware yet:

[image]

but that should serve as a starting point for additional variants and expanding the hardware.

vintagepc, 2020-08-15:
BTW- I'll add endstop outputs to the driver sim, makes the most sense to have them there.

3d-gussner, 2020-08-15:
Wow you are fast

3d-gussner, 2020-08-15:
I use muniRAMBo_1_0a instead of 1.1b as this is the base in Prusa Firmware.
I couldn't find any schematics for the 1.0a version so I stuck with the Prusa Firmware pins file.

3d-gussner, 2020-08-15:
Does it makes sense to add the MK42 and MK52 beds as components as these are different.
- MK42 is 12V and has 9 calibration points at slightly different locations compared to the MK52
- MK52 is available in 12 and 24V

vintagepc, 2020-08-15:
> I use muniRAMBo_1_0a instead of 1.1b as this is the base in Prusa Firmware.
> I couldn't find any schematics for the 1.0a version so I stuck with the Prusa Firmware pins file.

Ah, ok. I just renamed the header because the internal class name differed from the filename, and d I saw the notes about 1.0 not having LCD headers. (so I assumed the filename was what was incorrect). Feel free to correct.

> Does it makes sense to add the MK42 and MK52 beds as components as these are different.
> 
> * MK42 is 12V and has 9 calibration points at slightly different locations compared to the MK52
> * MK52 is available in 12 and 24V

I don't think the bed voltages impact the simulation and necessitates separate parts - but we would want to account for this in the PINDA, where the calibration magic happens. I believe it already has over-loadable functions and input arguments to configure how the calibration mapping happens. Only change is that for these beds, the sheet is permanently "off", since they are just PEI straight on the PCB.

Working on the stepper driver now :+1:  This thing is so much simpler than the TMC :rofl:

vintagepc, 2020-08-15:
One driver sim pushed :tada:

vintagepc, 2020-08-15:
... Or not, just added the thermistors and it doesn't step. womp womp :rofl:

3d-gussner, 2020-08-15:
The existing PINDA could be renamed to PINDAv2 and for the MK2 PINDAv1 we just exclude the thermistor, or not?

vintagepc, 2020-08-15:
You could just use the existing one as-is, the thermistor isn't built into the fake PINDA

3d-gussner, 2020-08-15:
SIM PINDA is better than fake😉

vintagepc, 2020-08-15:
sounds like you've had to deal with some fake ones in the wild :smile:

vintagepc, 2020-08-15:
d'oh... I have the pins backwards... driver inputs were connected as outputs. 

You're right, fake parts are terrible! :stuck_out_tongue_closed_eyes:

vintagepc, 2020-08-29:
I've pulled in improvements from master. 

... How does the mk2.5 fan check work? The fans are two wire, yet the firmware has a TACH pin defined that runs to pin 30 (X MAX)

3d-gussner, 2020-08-30:
Sorry, was busy with other things and you found it.

vintagepc, 2020-08-30:
Seems to work for the most part but XYZ cal fails with no obvious error (progresses through all 4 points)

vintagepc, 2020-09-14:
Fixed XY cal, added buzzer and PINDA thermistor. 

I believe the updates/models are as follows, correct?

MK2/S->MK2.5: Heatbed, noctua fans, PAT laser sensor, PINDA upgrade. 
MK2.5->S: Extruder parts + IR sensor

Alas there doesn't seem to be historical info as to the MK2->/S variant.

leptun, 2020-09-15:
Mk2 to S upgrade I think was just with better rods and U-bolts for the bed instead of zip-ties

vintagepc, 2020-09-15:
That would explain why I couldn't find anything on the electronics changes :)

vintagepc, 2020-09-15:
Apparently the 360 views on the site can be exported to F360 (and from there to obj/mtl). I don't think it's the case, but were there ever such pages for the MK2 line?

vintagepc, 2020-09-16:
I got a little too deep into this last night... :rofl: 

[image]

[image]

Pushed up initial files for the full compliment of miniRambo 1.3a based printers. Yes, that includes the MK1 and MK2MMU. 

I know the MMU is not quite right because it doesn't handle the E motors turning in opposite directions; I think everything else is at least wired correctly (in theory). MK2 XY cal is also still janky (fails correction) but it finds all the points. 

From the virtual hardware perspective:
- MK1 Z steps/mm = 4000
- MK2 = MK1 but with proper leadscrew motors. 9-point bed, no sheet. 
- MK2.5 -> Connects PAT, E-fan tach, PINDA temp, sheet, 4 point bed. 
- MK2.5S -> Swaps PAT for IR sensor.

leptun, 2020-09-16:
I think the MK1 has the old red bed with glass on top. Not sure about dimensions, but they might be different

vintagepc, 2020-09-16:
They are, I just didn't note that above :+1:

vintagepc, 2020-09-17:
MK2 XYcal fixed, MMU1 implemented better.

vintagepc, 2020-09-19:
I think I'm going to merge these printers in as-is and leave the 3D visuals for a separate issue, they're far enough along I don't think we want to hold them back for that. 

Going to start working on tests for the new components and printers with a draft PR to follow.

closed 2020-09-26 by vintagepc

vintagepc, 2020-09-26:
New follow-on issue for the 3d components is #243

PR #242 (2020-09-26): Add miniRambo printers (MK1 and MK2 generation)
### Description

Adds MiniRambo printers to the system. No 3D graphics yet (OBJ/MTLs TBD)

### Behaviour/ Breaking changes

Adding new features.

**NOTE: This PR changes the default "storage" .bin files so that each unique printer has its own set of storage files! e.g [Printer]_SDCard.bin, [Printer]_eeprom.bin, etc**

### Have you tested the changes?

Added tests for components, bootup of new printers

### Other

3D Graphics TBD in a separate issue/PR.

### Linked issues:

 - closes #219