vintagepc/MK404 #220: [BUG] gitversion: Clean ubutntu install does not have python but has python3

bozzzzo, 2020-08-11:
**Describe the bug**
`cmake` fails with failing to execute `python`

**To Reproduce**
1. cmake ..
```imavr binary dir detected as x86_64-linux-gnu
Defining simavr location as /home/foo/Projects/MK404/3rdParty/simavr/simavr/obj-x86_64-linux-gnu/libsimavr.a
/usr/bin/env: ‘python’: No such file or directory
CMake Error at 3rdParty/gitversion/cmake.cmake:18 (MESSAGE):
  Error running messmer/git-version tool.  Return code is: 127
Call Stack (most recent call first):
  3rdParty/gitversion/cmake.cmake:39 (_CREATE_GIT_VERSION_FILE)
  CMakeLists.txt:217 (TARGET_GIT_VERSION_INIT)

-- Configuring incomplete, errors occurred!
```
**Expected behavior**
A working build script

**Desktop (please complete the following information):**
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal

I made a PR https://github.com/smessmer/gitversion/pull/9

vintagepc, 2020-08-11:
Nice. We'd probably have hit this once the ubuntu-latest buildrunner becomes 20.04LTS (it's currently still 18). 
We'll pull that in once its updated upstream.

bozzzzo, 2020-08-12:
I find it good practice to fork all submodule projects. This allows fixing small stuff like this
before the upstream reacts.

wavexx, 2020-08-13:
Hit this as well on debian unstable... uurgh.

vintagepc, 2020-08-13:
yeah, looking deeper upstream seems to be stale. I'll fork/merge/whatever at some point.

PR #224 (2020-08-13): Misc cleanup, Python3
### Description

Fork and use a modified gitversion that works with systems sans python but with python3

### Linked issues:

 - fixes #220
closed 2020-08-13 by vintagepc