vintagepc/MK404 #223: [BUG] MMU motors moving too fast

vintagepc, 2020-08-12:
**Describe the bug**

The selector (and possibly other motors) on the MMU are moving too fast. Git bisect shows this is result of changes for #198 

```
git bisect bad
d3f6cc44fcb10f93e671f52c30d03b0d614195da is the first bad commit
```

commit d3f6cc44fcb10f93e671f52c30d03b0d614195da
Author: @wavexx 
Date:   Tue Aug 4 13:28:49 2020 +0200

    Implement TMC2130 variable resolution support (#198)
    
    - Also handle CHOPCONF.mres.
    - Introduce helpers to convert from/to step/position
    - Update limits as CHOPCONF is updated.

**To Reproduce**

1. Run with MMU
2. Choose 'load all filaments to make it home the selector
3. Observe selector motion

**Expected behavior**
Selector should not move nearly as quickly (looks like it's taking steps/mm as lower than it should be)

wavexx, 2020-08-13:
On Thu, Aug 13 2020, vintagepc wrote:
> **Describe the bug**
>
> The selector (and possibly other motors) on the MMU are moving too
> fast. Git bisect shows this is result of changes for #198

I assigned the default resolution to 16 usteps to match xyz on the MK3,
which is probably wrong for the others.

I'll have a look later on.

leptun, 2020-08-13:
@wavexx 
```
	case AX_PUL: return tmc2130_usteps2mres((uint16_t)2);
	case AX_SEL: return tmc2130_usteps2mres((uint16_t)2);
	case AX_IDL: return tmc2130_usteps2mres((uint16_t)16);
```

vintagepc, 2020-08-13:
yep, looks like that would explain it ;) Idler behaves fine, Id suspect the pulley also feeds super quickly now too :rofl: 
I guess that's one way to speed up your toolchanges....

closed 2020-08-29 by vintagepc

wavexx, 2020-08-29:
On Sat, Aug 29 2020, vintagepc wrote:
> Closed #223 via #228.

Damn, feel bad for not having worked on this!
But the fix is better than what I had in mind =)

vintagepc, 2020-08-29:
NP, it happens :)

I took some inspiration from the way we handle this for the A4982s.

PR #228 (2020-08-29): Improve MRES implementation
### Description

Changes the chopconf MRES handling to be easier to understand

### Behaviour/ Breaking changes

MRES no longer alters the bounds of the motor. Instead, we always work in the "maximum" microstep resolution of the driver, and increment the internal step count by 2^mres.

Motor steps per MM for TMC drivers is now input as the steps/mm if the driver was operating at full resolution, not the Marlin/M92 value.

### Have you tested the changes?

MMU now behaves as expected, main printer still moves as it should and the TMC automated test still works as exp [... cut]