vintagepc/MK404 #230: [FEATURE] Expand testing

vintagepc, 2020-08-30:
**Describe the solution you'd like**
Expand the automated testing to cover more of the code. 

Just start simple with something that runs on a system with GL, deal with the headless build runners later... PPM seems like a nice convenient format.

It should:

- exercise the GL draw code
AND
- take snapshots?
- Compare them to a reference?
OR
- Inspect key pixels to make sure the visual looks like it should. May be easier with some drawables than others. 

**Additional context**
Useful links for automated GL testing...
https://github.com/openrndr/setup-opengl

https://github.com/cirosantilli/cpp-cheat/blob/70b22ac36f92e93c94f951edb8b5af7947546525/opengl/offscreen.c

https://www.khronos.org/registry/OpenGL-Refpages/gl4/html/glReadPixels.xhtml

https://github.com/mesa3d/piglit

http://pdiff.sourceforge.net/

vintagepc, 2020-09-02:
Looks like EGL is required to run headless on the build server without opening a window... GLUT needs one.

vintagepc, 2020-09-12:
@DRracer This one (and the upcoming associated PR, #231) should be of particular interest to you, it adds and makes use of _extensive_ scripting and verification capability for components, telemetry, and tests. So while it serves as unit tests for the simulator, it should be more than suited to automated evaluation of firmware behaviour prior to release!

DRracer, 2020-09-12:
@vintagepc well done, making a note for the upcoming FW release - we'll try it. Thanks for notifying me.

PR #231 (2020-09-13): #230 expand testing
### Description

Adds functionality to verify, and some tests for, the GL componennts

### Other

New scripting feature to snapshot the current framebuffer

### Linked issues:

 - closes #230
closed 2020-09-13 by vintagepc