vintagepc/MK404 #232: [BUG] Mac binary (0.9.4) fails to launch

thinkyhead, 2020-09-01:
**Describe the bug**

When attempting to launch the macOS binary the following error is produced:
```
dyld: Library not loaded: /usr/local/opt/glew/lib/libGLEW.2.1.dylib
  Referenced from: /path/to/MK404-v0.9.4/MK404_OSX
  Reason: image not found
Abort trap: 6
```

The `/usr/local/opt/glew/lib` path seems pretty strange.

**To Reproduce**

1. Open Terminal and `cd` to the `MK404-v0.9.4` folder
2. Use `chmod ug+x MK404_OSX` to make the macOS binary executable.
3. Right click on `MK404_OSX` in the Finder and choose 'Open'
4. Authorize the binary to run on the system.
5. Go back to Terminal and run the program from there… `./MK404_OSX`
6. You may see the error detailed above.

**Expected behavior**
The binary should find my installed glew library at `/opt/local/lib/libGLEW.2.1.dylib`, or it should be built and packaged as an `.app` using XCode, with required third-party libraries installed in the application bundle.

**Desktop:**
 - MacOS Catalina 10.15.6

**Additional context**
I'm using the MacPorts package manager to manage optional packages in the `/opt/local` folder.

vintagepc, 2020-09-01:
Thanks for the report. I suspect that may be a difference between our build runner config and your machine having the libraries in different paths.

Alas I do not have a machine to test and debug changes or set up .app packaging, so what I can do to resolve this is severely limited.

thinkyhead, 2020-09-28:
The `/opt/local` folder is used by the MacPorts package manager, and possibly others. Homebrew may also use `/opt/local`. Custom installs can go into `/usr/local` as with other versions of BSD UNIX.

I don't use HomeBrew. I do use MacPorts.

However, the correct way to install dependencies on macOS is to compile shared code down to an `.a` library file and optionally package the library the form of a `.framework` bundle. When making a macOS binary for distribution to common users, it is best to use Xcode and set it up to produce an `.app` bundle with the dependent libraries and frameworks installed within the application bundle's `Resources` folder so that there are no external (non-system) dependencies that need installation.

vintagepc, 2020-09-28:
Thanks for the explanation/clarifications. Hopefully it will be possible to address this in the future. :+1: