vintagepc/MK404 #233: Failing XYZ calibration[BUG?]

pkg2000, 2020-09-07:
Good day, and thank you for reading. Awesome project, but I am having some issues. After getting it up and running, clearing the Selftest, the XYZ calibrations are consistently failing on the 1 of 4. After a little bit of spiraling, I get the following lines (screenshot below) saying something about the buffer, and then it starts feeding me very specific nonsense (4848484848484848.....). Been looking around the wiki if there is some config for buffer or whatever, but can't find anything regarding this.

Steps to reproduce the behavior:
Clean VM install of MXLinux or Ubuntu
Find them deps.
compile
run

**Expected behavior**
Successful calibration.

[image]

**Machine:**
 - Windows 10 running both MXlinux and Ubuntu through VMware Player 15, non-commercial.
 - Intel core i5 laptop and fx-8350 R9-270 rig.

Tried different ways of starting it, but this is basically what I execute with. Nohack (-n) just in that case.
[image]

Following your instructions and the struggle I had doing so, tells me that you guys are a few levels beyond me, but I am not a complete noob. I think you guys have done an amazing job with this and I would greatly appreciate any feedback you could provide on this issue. I imagine (and hoping) that there is something obvious I am missing here. I'd also be happy to write something up for your documentation, once I get going properly.

Thank you!

vintagepc, 2020-09-07:
Hi here!

Sounds like a simple oversight - it looks like you forgot to "remove" the virtual steel sheet before doing the calibration. 

You can do this by pushing the `y` key, or using the middle-click context menu. 

(The wall of `48`s is a visual representation of what the PINDA is seeing, when the sheet is on it's going to be mostly level, but if the sheet is off, it should be mostly zeros with a circular group of higher values in the center that represent the bed's calibration point). 

Hope that helps!

leptun, 2020-09-07:
The `sizeof()` are not important. It's simply debug info that in my opinion is absolutely useless...

As for the 484848484848.... That is the level at which the probe triggered when it was scanning for the bed point.
It can mean 2 things:
- You forgot to "remove the sheet from the bed", hence why it's so level and without the point to find. Keybind is `Y`
- Some regression in code

vintagepc, 2020-09-07:
Not ruling out a regression entirely, but I do know it was working recently while I was busy on the MK2.5 implementation :)

leptun, 2020-09-07:
That's why I put the "forgot to remove sheet" first. :)
The timing of your comment was perfect. I commented and then yours appeared on top of mine

vintagepc, 2020-09-07:
And now you know why my avatar is a ninja tux :rofl:

pkg2000, 2020-09-07:
Aaah. Thank you for all of your responses. That makes very much sense from my experience with the IRL machine. Hit "no" when it asked, but I of course have to take it off as well;) Currently not using the fancy graphics either as it ate the little cpu I could give the VM. Thank you. Will confirm in a bit.

vintagepc, 2020-09-07:
You might have more luck with `-g lite` which renders only the minimal functional parts - bed + sheet, nozzle, fans,  and a few other components. 
It shouldn't perform too badly with even minimal software rendering - but you might want to double check the graphics settings. Many VM tools offer the ability to assign graphics memory and limited 3d acceleration support. 

While not actively maintained, it should compile and run on cygwin, which would get you the ability to run on your native graphics hardware.

pkg2000, 2020-09-07:
Yeah, was indeed the sheet. Thank you! I will give -g lite a go, and also try to tick of 3d hardware (was an option), but this laptop is part potato anyhow. Will give cygwin a try, but wanted to run it in what you guys were running and read your disclaimer regarding other platforms. Again, thanks a bunch!

vintagepc, 2020-09-07:
Excellent, good to hear you're all set. 

(FWIW I'm on a laptop from 2013 and using the integrated graphics for my dev activities)

closed 2020-09-07 by vintagepc