vintagepc/MK404 #235: [Q+A] Build documentation

leptun, 2020-09-12:
Not sure if I'm just blind, but I was not able to see which packages were needed for a complete build from scratch. I had to go to the build task for github to see exactly which packages needed to be installed. In any case, is there any documentation on what is needed? I know that at some point it was in a special file in the root of this repository, but now it's gone.

Btw, do you use vscode for this project? I personally do and I think it would be beneficial if we added a list of recommended extensions that can be installed when you open the project. https://code.visualstudio.com/docs/editor/extension-gallery#_workspace-recommended-extensions

vintagepc, 2020-09-12:
Yes - I moved the various READMEs and other details to the project Wiki:

https://github.com/vintagepc/MK404/wiki/Supported-Operating-Systems

I do use VSCode too, but really only with the cmake and C++ extensions. But there's definitely a case for that if we find more obscure but useful extensions for working on the project.

leptun, 2020-09-12:
Ok. Thanks for letting me know where it was moved. Closing this now

closed 2020-09-12 by leptun

vintagepc, 2020-09-12:
No problem