vintagepc/MK404 #236: [BUG] Backlight regression

leptun, 2020-09-13:
**Describe the bug**
The backlight colouring is broken. Not sure if it is a rendering issue or if the wrong values are extracted from the timer output.
Was there any recent change to the timers? I think you added an unsupported mode recently.

**To Reproduce**
Steps to reproduce the behavior:
Set backlight in firmware to different values and enhoy the show

**Screenshots**
[image]
video with many more values: https://photos.app.goo.gl/4K3ra2awqQ7THUhs7

**Desktop (please complete the following information):**
 - OS: Ubuntu desktop 20.04 - amd64
- MK404: latest master as of writing (commit fa555a28d72390399381b9508c2db46ac0a0b08f)

**Additional context**
Not sure if it is firmware, but I'm confident it should never be a shade of green :)

vintagepc, 2020-09-13:
Huh, that's a weird one. This backlight breaking seems to be becoming rather trite in this project :laughing: 

I remember seeing something similar but with red after I had been changing the glClearColor to debug the "black screen on nVidia cards" issue. So I think it's a transparency/alpha problem and we're just showing whatever happens to be behind the display.

leptun, 2020-09-13:
This is on a different computer than with my previous setup. It’s very possible to be an nvidia only issue. This is the only card in the system.
On the other one I have both integrted graphics (main) and dedicated nvidia.

Not sure if it influences anything, but I also use a different ubuntu version than on the other setup.

vintagepc, 2020-09-13:
Nope, I found the issue, I messed up the colour scaling calculation and was dividing by the PWM (0-1.0f) instead of multiplying by it to get the new RGBA quad. 

You get a very interesting occurrence if you increment it by values of one and it changes drastically from screen to screen XD

commit 07b0a253 (2020-09-13): Fix #236, check backlight test
commit 4f4b3b43 (2020-09-13): #230 expand testing (#231)

* add a GL testing helper
* add GL test functionality, extensive tests
* Fix #236
closed 2020-09-13 by vintagepc