vintagepc/MK404 #243: [ENHANCEMENT] 3D visuals for miniRambo printers

vintagepc, 2020-09-26:
This is a continuation of #219. 

Now that the base printers are in, we will need to source and add some models for them. We might end up needing to model them ourselves in a CAD Program using the part STLs and other components.

3d-gussner, 2020-10-18:
What do we need to get this finished?
As we are investigating the VFA it might be interesting to compare it to MK2 and MK2.5/s.

3d-gussner, 2020-10-18:
I have been trying to run the Wizard on:
- MK2 works fine and finishes
- MK2.5+S failed at XYZ cal ... have to to retest it again

vintagepc, 2020-10-18:
For the models we'd need something we can convert to OBJ via known tools. Essentially some sort of CAD assembly. I did not see any public ones out there like the one we used for the MK3. 

We'd either need to build one ourselves from the part STLs and hardware or maybe Prusa has some assembly models they'd be willing to share - but I am not sure since the earlier ones were heavily designed in OpenSCAD and they only moved to Autodesk later. 

I suppose there's no reason we couldn't use the MK3 models for the same if we just want the 3d visuals, it'd just look a little strange. 

I did push up a fix for the 2.5S - the Z height hadn't been corrected to allow for the extra 5mm over the 2.5.

3d-gussner, 2020-10-18:
I have a look if I still have a MK2 and MK2.5/S in fusion

3d-gussner, 2020-10-19:
First test looking good. Thanks

vintagepc, 2020-10-20:
BTW if you post the STEP files or email them to me I may have some time to knock this out.

vintagepc, 2020-10-24:
FWIW I did find a MK2 STEP file online and can work with that instead if necessary.

vintagepc, 2020-10-27:
Just pushed up an initial commit. 

TODOs:

- [ ] Add print bed for 2.5 & S variants
- [ ] Add MMU parts for MK2MMU variant.

vintagepc, 2020-10-27:
additional push for two of three. Doing the MMU parts is going to be trickier, I have the STLs, I just need to learn more about SolidEdge to manipulate them as parts and get them placed in the STEP file. I may make that a separate issue.

3d-gussner, 2020-10-28:
I will have at look for the MMUv1 parts.
BTW the MK2 fancy looks great.
The MK2.5 needs a different Extruder.

vintagepc, 2020-10-28:
Whoops, yes. I forgot about that >_<

TODO list edited ;)

3d-gussner, 2020-10-28:
Started to convert MMUv1 from SCAD to STEP and OBJ, MTL using FreeCad 0.19 and 3d-builder.
https://github.com/3d-gussner/Original-Prusa-i3/tree/Multi-Material-Upgrade/Printed-Parts
Please let me know if these are okay or I have to clean these a bit?
In the past the conversion from SCAD to STEP via FreeCad 0.1? and Fusion360 needed some cosmetic cleanup.

vintagepc, 2020-10-28:
Thanks! I'll have a look today.

vintagepc, 2020-10-28:
Yes, if you could do the switchbox v2 and cover v2 I think it'll be a nice touch. 

STEP alone is fine, I can manipulate them into place and export as a single group rather than having to handle the individual OBJs in MK404 😄 

This is forcing me to learn SolidEdge, and I'm starting to like it more and more, apart from the fact it doesn't run in Wine/Crossover.

3d-gussner, 2020-10-28:
Done https://github.com/3d-gussner/Original-Prusa-i3/tree/Multi-Material-Upgrade/Printed-Parts/STEP
Cleanest is the fusion step file

vintagepc, 2020-10-28:
Getting there! I have the MMU models in. I will still need to think about how to do visualizations for the E motors and prints though; the current MMU is somewhat baked in at a lower level so I'll have to decouple that somehow and route connections appropriately. 

[image]

vintagepc, 2020-10-29:
Well that was a pleasant surprise! Just connecting up the motors (and fixing a colour bug with the additional virtual prints)... it seems to be working!

[image]

I think I'll get some tests written, and then merge this work in and leave the E visualizers for another day when someone actually needs them to debug something :smile:

PR #279 (2020-10-29): #243 - MK2 visuals
### Description

Adds 3D visuals for the MK2-, -MMU, and 2.5x series printers. 

### Behaviour/ Breaking changes

Adds visuals - should not break anything else. 

### Have you tested the changes?

Will add test cases to start these printers and snapshot the renderings. 

### Other

- Fixed a bug with GLPrint colours always being red
- Adds GLObj support for OBJs/MTLs that do not set `d` and/or have reverse triangle winding. 
- No E-vis on the motors for the MK2MMU (Future task when needed)
- MK2 "lite" visual uses MK3 lite instead.

### Linked issues:

- Fixes #243
closed 2020-10-29 by vintagepc