vintagepc/MK404 #247: [BUG] simavr cmake integration suboptimal

wavexx, 2020-09-27:
**Describe the bug**
Perhaps unexpectedly simavr still builds object files inside the source tree (3rdParty/simavr/obj-*) even when building off-tree.

Not a huge problem, but the main gotcha is that a build can become stale in a few situations.
Also, the `clean` target doesn't actually remove these.

**To Reproduce**
Run cmake outside of the source tree and make.
A subdirectory will suffice:

````
cd MK3SIM
mkdir build && cd build
cmake ..
make
````

vintagepc, 2020-09-27:
The in-source building is an explicit external_project config: 
https://cmake.org/cmake/help/latest/module/ExternalProject.html

I enabled that for one reason or another - I think it had something to do with the way it generates its object structure. 
But we do also have BUILD_ALWAYS enabled, which should prevent update issues:
```

BUILD_IN_SOURCE <bool>
When this option is enabled, the build will be done directly within the external project’s source tree. This should generally be avoided, the use of a separate build directory is usually preferred, but it can be useful when the external project assumes an in-source build. The BINARY_DIR option should not be specified if building in-source.

BUILD_ALWAYS <bool>
Enabling this option forces the build step to always be run. This can be the easiest way to robustly ensure that the external project’s own build dependencies are evaluated rather than relying on the default success timestamp-based method. This option is not normally needed unless developers are expected to modify something the external project’s build depends on in a way that is not detectable via the step target dependencies (e.g. SOURCE_DIR is used without a download method and developers might modify the sources in SOURCE_DIR).
```

```
ExternalProject_Add(simavr
    SOURCE_DIR ${PROJECT_SOURCE_DIR}/3rdParty/simavr
    CONFIGURE_COMMAND ""
    BUILD_COMMAND make build-simavr RELEASE=0
    INSTALL_COMMAND ""
    BUILD_IN_SOURCE 1
    BUILD_ALWAYS 1
)
```

I suppose the better thing to do is make a CMakeLists.txt file for SimAVR so we can just do an include_subdirectory rather than relying on SimAVR's own build system.

vintagepc, 2020-09-28:
Yes - I just checked, simavr doesn't build if we try to build out-of-source. 

Would it help alleviate the issue if I add a custom convenience target to clean simavr? 

e.g. `make clean-simavr` and/or `make clean-all` ?

closed 2020-09-29 by vintagepc

wavexx, 2020-09-29:
On Mon, Sep 28 2020, vintagepc wrote:
> Would it help alleviate the issue if I add a custom convenience target to clean simavr?

That would be more than enough IMHO.
I think adding custom a simavr build is overkill.

vintagepc, 2020-09-29:
Great- that should already be in master now