vintagepc/MK404 #248: [BUG] PAT9125 not detected in hardware I2C mode

wavexx, 2020-09-27:
**Describe the bug**
The SCL/SDA irqs do not seem to be generated when the PAT9125 is used with the HW TWI, causing the simulated PAT9125 to be non-responsive.

**To Reproduce**
Attaching an MK3 FW with https://github.com/prusa3d/Prusa-Firmware/pull/2814 merged:

[Firmware.ino.zip](https://github.com/vintagepc/MK404/files/5288227/Firmware.ino.zip)

vintagepc, 2020-09-27:
Derp. Someone even left me a nice TODO... :rofl: 
```

	void _OnI2CTx(avr_irq_t */*irq*/, I2CMsg_t /*msg*/)
		{
			// TODO(vintagepc) - I don't have any real "hardware" i2c items to simulate at the moment.
		}
```

wavexx, 2020-09-27:
I noticed that, but I'm curious now. shouldn't the SDA/SCL pins still trigger an irq irregardless?
I guess the TWI module doesn't do that.

vintagepc, 2020-09-27:
Yeah, SimAVR doesn't do that for most cases - for example, the digital pin IRQ does not toggle if it is in PWM mode. 

I just need to wire up the official I2C IRQs to the message receiver and call the shared get/set reg value methods.

vintagepc, 2020-09-27:
Hmm... not quite so simple. I've prototyped it in and get the message/send an ACK but it doesn't appear to make it through to the firmware - it never attempts to read the device register to get the PID/VID.

wavexx, 2020-09-27:
On Sun, Sep 27 2020, vintagepc wrote:
> Hmm... not quite so simple. I've prototyped it in and get the
> message/send an ACK but it doesn't appear to make it through to the
> firmware - it never attempts to read the device register to get the
> PID/VID.

Is there something I can build for testing too?

vintagepc, 2020-09-27:
So far I see the TWI library in the Prusa FW is expecting to see TW_MT_SLA_ACK (0x18) but based on code inspection/stepping the status register is ending up at 0x28 (TW_MT_DATA_ACK). Not sure why yet. A build that dumps TWI status over serial might be useful to figure out what the firmware is seeing vs what it expects. 

If you'd like to take a look or poke at this, I pushed up https://github.com/vintagepc/MK404/tree/248-Hardware-TWI that has the basic framework outlined in SPIPerpihheral.h for "real" hardware SPI.
Once we get the ACK to work and the firmware to actually query with a `TWI_COND_READ` or address set it should be just a matter of adding in some code to call the Set or GetRegVal functions. 

Just let me know if you're interested in taking a crack; then I can focus on other tasks so we're not both doing the same work :+1:

wavexx, 2020-09-27:
On Sun, Sep 27 2020, vintagepc wrote:
> So far I see the TWI library in the Prusa FW is expecting to see
> TW_MT_SLA_ACK (0x18) but based on code inspection/stepping the status
> register is ending up at 0x28 (TW_MT_DATA_ACK). Not sure why [...]

So this stops already at the first read of the PID1 register when the FW
is about to transmit the register address.

> Just let me know if you're interested in taking a crack; then I can
> focus on other tasks so we're not both doing the same work :+1:

Although I'm running this on the real thing, I don't exclude FW quirks
here as well which is why I'm all ears :P

I don't think I can do this today anymore though

vintagepc, 2020-09-27:
Of course; I know we all have different amounts of time we can commit to the project. 

I'll keep looking.

wavexx, 2020-09-27:
On Sun, Sep 27 2020, vintagepc wrote:
> Of course; I know we all have different amounts of time we can commit
> to the project.

Mostly because I'm chasing another bug already =)

vintagepc, 2020-09-27:
aha, it's a similar issue to what we had with the SD card's SPI bus. 
The simulation doesn't handle faster clock rates, so by the time the status register is updated, we've already checked it in `pat_probe` and it's too late. 

```
/*
 * This is supposed to trigger a timer whose duration is a multiple
 * of 'twi' clock cycles, which should be derived from the prescaler
 * (100khz, 400khz etc).
 * Right now it cheats and uses one twi cycle == one usec.
 */
```

wavexx, 2020-09-27:
On Sun, Sep 27 2020, vintagepc wrote:
> aha, it's a similar issue to what we had with the SD card's SPI bus.
> The simulation doesn't handle faster clock rates, so by the time the
> status register is updated, we've already checked it in `pat_probe`
> and it's too late.

Ha. So it should work if I reduce the twi clock speed to 100khz for
example? That's easy to test.

vintagepc, 2020-09-27:
In theory - but there also seems to be a read/write handling problem in the ACK types. I have some local modifications to do the same and while it now communicates something is still wrong with the writes.

vintagepc, 2020-09-27:
... so this doesn't make sense. 
in PAT9125.c when doing a write:

```
#elif defined(PAT9125_I2C)
	if (twi_rw8(PAT9125_I2C_ADDR,TW_WRITE,&addr) ||
        twi_rw8(PAT9125_I2C_ADDR,TW_READ,&data))
```

.. you're not actually providing any data to be written.

vintagepc, 2020-09-28:
Can confirm HW TWI works if hacked around, but blocked pending a clean resolution in https://github.com/buserror/simavr/issues/137

(and the associated upstream PR, https://github.com/buserror/simavr/pull/401

vintagepc, 2020-09-29:
I'm picking in the changes that work for us; will re-visit once the upstream discussion is resolved or we have additional hardware TWI components that indicate the fix is flawed.

PR #251 (2020-09-30): #248 hardware TWI
### Description

Adds support for hardware-based TWI and the same on the PAT9125 sensor.

### Behaviour/ Breaking changes

SimAVR update required; added functionality, might have broken something related to hardware TWI, but as we have no other hardware TWI components, I don't know. 

### Have you tested the changes?

Verified against @wavexx 's hardware-TWI firmware. 

### Other

"Proper" resolution of TWINT still pending with the upstream TWI fixes. This works well enough for us. 

### Linked issues:

- Closes #248
closed 2020-09-30 by vintagepc