vintagepc/MK404 #249: [BUG] SD target broken

leptun, 2020-09-28:
I think the sd card target is currently broken in the latest master (or at the very least the latest master this morning). It generates an outdated output filename for the sd card image file.
Instead of Prusa_MK3S_SDcard.bin it generates Einsy_mega2560_sdcard.bin (or something like that. Writing these names from memory). I have to manually rename the generated file for the specific printer hardware config I’m using.

vintagepc, 2020-09-28:
Oops, yeah... forgot about that when I changed it so each printer has its own SD card and storage files.

vintagepc, 2020-09-29:
[image]

PR pending, I've got it auto-generating a list of printer SD card targets in CMake now.

3d-gussner, 2020-09-30:
Still broken as it builds the SD cards but with wrong name:
`SD card image (Prusa_MK25S_13_SDcard.bin) could not be mounted (error -1 ).`
but it created `Prusa_MK25S_mR13_SDcard.bin`

vintagepc, 2020-09-30:
shoot, the minirambo naming throws a wrench in the works there since the printer name does not reflect the c++ class name used to get the storage filenames.

I'll fix tonight.

leptun, 2020-10-05:
The saga continues.
Added a gcode file to SDCard named `Buton_umbrela_Florica_MKB_FAST.gcode`, ran the Prusa_MK3S_SDcard.bin target and got this in the build log:
```
[build] Wrote 134217728 bytes to SD image.Wrote /home/alex/Desktop/MK404/build/Prusa_MK3S_SDcard.bin. You can now use mcopy to copy gcode files into the image.
[build] Copying Test.gcode
[build] [100%] Built target Prusa_MK3S_SDcard.bin
[build] Build finished with exit code 0
```
As can be seen, only Test.gcode was copied.

leptun, 2020-10-05:
Ok. Good to know. Just did this and it works.
One more issue, though. Files that have "()" break the generated makefile. Something to look into.
```
[build] Wrote 134217728 bytes to SD image.Wrote /home/alex/Desktop/MK404/build/Prusa_MK3S_SDcard.bin. You can now use mcopy to copy gcode files into the image.
[build] /bin/sh: 1: Syntax error: "(" unexpected
[build] make[3]: *** [SDCard/CMakeFiles/Prusa_MK3S_SDcard.bin.dir/build.make:58: SDCard/CMakeFiles/Prusa_MK3S_SDcard.bin] Error 2
[build] make[2]: *** [CMakeFiles/Makefile2:2265: SDCard/CMakeFiles/Prusa_MK3S_SDcard.bin.dir/all] Error 2
[build] make[1]: *** [CMakeFiles/Makefile2:2272: SDCard/CMakeFiles/Prusa_MK3S_SDcard.bin.dir/rule] Error 2
[build] make: *** [Makefile:1137: Prusa_MK3S_SDcard.bin] Error 2
[build] Build finished with exit code 2
```
Used filename was `Test (2).gcode`

vintagepc, 2020-10-05:
Good find. I probably just missed putting quotes around a filename variable somewhere.

vintagepc, 2020-10-05:
OK, I have a fix for both of these, I think. I managed to make the wildcard for `*.gcode` work, which both pulls in the current state of the directory and fixes the issue with special chars in the filename.

Cmake convenience entries for gcode files are still dependent on re-running cmake to pick up new ones to list under a target though.

vintagepc, 2020-10-05:
Third time's the charm, I hope.

PR #250 (2020-09-29): Add custom SD card targets (fixes #249)
### Description

Fixes stale SD card target in CMake

### Behaviour/ Breaking changes

N/A - adds more convenience targets

### Have you tested the changes?

Checked locally, it built an SD target and copied the gcode files into it.

### Linked issues:

 - Fixes #249
commit d75e8c82 (2020-09-29): Add custom SD card targets (fixes #249) (#250)
PR #254 (2020-10-01): Fix bed heaters on MK2.5 and SD target names
### Description

Fixes #249  (again)
Fixes #252 (MK2.5 bed heater stuck on)
closed 2020-10-05 by vintagepc