vintagepc/MK404 #253: [BUG] MK2 PINDA doesn't trigger

3d-gussner, 2020-09-30:
**Describe the bug**
Started Prusa_MK2 with latest official firmware and tried to run the wizard, but i fails in the selftest at z.
Tried G28W to home and the PINDA doesn't trigger.

**To Reproduce**
Steps to reproduce the behavior:
1. Start ./MK404 Prusa_MK2_mR13 -s -f (MK2 hex file)
2. Run selftest or `G28W`
3. Selftest fails and PINDA doesn't trigger

**Expected behavior**
Should trigger above the MK42 cal points.

**Desktop (please complete the following information):**
 - OS: [Ubuntu VM]
 - Version [18.04]

**Additional context**
Saw that you added the MK42 cal points but looks like it doesn't work.

vintagepc, 2020-09-30:
Interesting; I know I had the self test completing at one point. I wonder what changed...

vintagepc, 2020-09-30:
This is a weird one,  the microstep count (and therefore position tracking) doesn't seem to be working right. 

I can see the firmware is setting the driver MS1/2 to H/H (MS 16), correct according to the source.

But when I try a move axis/step, it's only stepping about 1/16th of what it should. Are the MS outputs inverted? Doesn't look that way on the schematic.

vintagepc, 2020-10-02:
Any chance @wavexx  or @leptun has some insight on the miniRambo line with regards to this? From what I can see it looks like the FW is stepping as though it's in full-step mode but the driver is configured for 16ths. 

Steps/mm should be configured correctly as well.

wavexx, 2020-10-02:
No direct experience with the miniRambo

PR #267 (2020-10-09): Patch A4982 step-position conversion functions
### Description
Fixes steps/mm being calculated incorrectly on miniRambo based boards.

### Behaviour/ Breaking changes
None

### Have you tested the changes?
Tested Prusa_MK2_mR13 variant with two different microstepping configurations. Both worked as expected. 

### Linked issues:
closes #253 

[image]
[image]
closed 2020-10-09 by vintagepc