vintagepc/MK404 #266: [BUG] MBL map does not appear to work for print simulation

vintagepc, 2020-10-09:
**Describe the bug**

Per the discussion on #264, it looks like the printer MBL map isn't working right. Investigate.

DRracer, 2020-10-09:
Update - the white print line follows the bed (had to make a really bad heatbed with 1.5mm peaks), but the highres extrusion visualization doesn't (this lays flat onto the bed).

DRracer, 2020-10-09:
Screenshot from the sim:
[image]
notice the white line going up on the right side

vintagepc, 2020-10-09:
Ah, this may just be a simple oversight. It looks like AddSegment() uses the same Z start/end values for the extrusion chunk.

3d-gussner, 2020-10-09:
> Update - the white print line follows the bed (had to make a really bad heatbed with 1.5mm peaks), but the highres extrusion visualization doesn't (this lays flat onto the bed).

You should contact Support if you have that kind of bent bed 😁 and ask for replacement

vintagepc, 2020-10-09:
Sounds like the complaint is "my bed isn't broken enough, it prints too level!" to me :rofl:

DRracer, 2020-10-09:
@vintagepc I'm afraid this may get a little more complicated, because we need to take the Z-motor nonlinearity into consideration while processing the printed segment

vintagepc, 2020-10-09:
And the backlog gets longer.... :stuck_out_tongue: 

Are we thinking part of the cause of 602 is non-linearity coupled with MBL?

vintagepc, 2020-10-09:
I pushed up a print-sim-enhancements branch which has a partial for what I bashed out this morning. 

It brings a dT (delta cpu cycles) through to OnEStep and fixes the Z endpoint of print segments. Didn't have time to check it though so it might be horribly broken.

DRracer, 2020-10-09:
Nope, issue 602 is purely caused by nonlinearity of the E-motor.
But for the sake of completness, Z-axis and MBL related issues (line segmentation) are missing in our analysis so far.

Delta CPU cycles: that should be enough, I'll look into it.

vintagepc, 2020-10-09:
I'm hoping just fixing the segment start/end Z might actually be enough to make issues visible given the x/y segment sizes tend to be very small in HR modes anyway. But if not we can improve the calculations. 

I suspect it might be necessary to carry the deltaT down through to the addSegment function, but I'm not 100% sure. It might be perfectly satisfactory to calculate an adjusted E value based on that during the main onEStep calculations as well.

vintagepc, 2020-10-10:
OK - can confirm that the simulated print now follows the MBL curve as expected:

[image]

Z nonlinearity can be enabled and with the quads being so small I expect that it should reveal any issues as a gap between adjacent ones as opposed to a smooth transition. 

Is there anything else do do on this particular issue or is this satisfactory for now?

PR #270 (2020-10-15): Enhance nonlinearity aspects
### Description

3D print now follows the correct Z path based on MBL results:

(extreme example)
[image]

### Behaviour/ Breaking changes

This may affect the automatic squish factor layer height code. But looking at the output it seemed happy, at least with the test print. Things will probably get weird with extreme warping and lots of retractions/travel.

### Have you tested the changes?

Locally, see image. 

### Linked issues:

- Fixes #266 
- Has framework for #265 (delta T time value in CPU cycles)
closed 2020-10-15 by vintagepc