vintagepc/MK404 #287: [Q+A] Running in Windows subsystem on Linux

espr14, 2020-11-25:
I can build MK404 on Ubuntu 20 on WSL on ext4. WSL (compared to WSL2) does not have display:0 but I used VNC so I ran it on display:1. It has some problem with OpenGL. I'll try WSL2.

vintagepc, 2020-11-25:
What sort of issue are you encountering? (error messages, console output etc)? 

Have you tried with the "native" cygwin build?

3d-gussner, 2020-11-26:
@espr14 I had MK404 running in WSL2 and it kind of worked. Changed to Dual boot due to my "old" PC wasn't performing that well with MK404 in WSL2.
Also to install MK404 on "fresh" Ubuntu this might help:
```
sudo apt install -y libelf-dev gcc-7 gcc-avr libglew-dev freeglut3-dev libsdl-sound1.2-dev libpng-dev cmake build-essential lcov mtools
```

espr14, 2020-11-26:
sudo apt install git

vintagepc, 2020-11-26:
That should be fine for most purposes, though the VM likely will not have enough GPU grunt for `-g fancy` or any of the 3d extrusion options.

espr14, 2020-11-28:
`-g lite` runs ~8 fps.

vintagepc, 2020-11-28:
Have you enabled 2d/3d acceleration in Virtualbox and installed the drivers in the ubuntu guest?

vintagepc, 2020-11-28:
https://blogs.oracle.com/scoter/oracle-vm-virtualbox-6-3d-acceleration-for-ubuntu-1804-guest

closed 2021-03-07 by vintagepc