vintagepc/MK404 #289: [BUG] cmake variable incorrect

yash-fn, 2021-01-22:
Simple Build Error.

MK404 target requires 
set (LIBSIMAVR ${PROJECT_SOURCE_DIR}/3rdParty/simavr/simavr/obj-${SIMAVR_BIN_DIR}/libsimavr.a)
to be changed to 
set (LIBSIMAVR ${PROJECT_SOURCE_DIR}/3rdParty/simavr/simavr/obj-${SIMAVR_BIN_DIR}.2.0/libsimavr.a)

in order for it to build. 

I don't know cmake so I don't know how to change that .2.0 expression into a wildcard or regex.

My system: Latest MacOS running on macbook pro. All dependencies installed using homebrew. 
MacOS Big Sure 11.1

vintagepc, 2021-01-22:
Not sure where that is coming from, we use the same syntax as simavr to determine what the build obj directory is. 

What compiler are you using and what is the output of `[compiler] -dumpmachine`?

yash-fn, 2021-01-22:
gcc-10 (Homebrew GCC 10.2.0_2) 10.2.0

I do not know how to determine [compiler] -dumpmachine. 

All I know is that the project wouldn't compile until I added the .2.0 part. But perhaps I did something wrong. If you are sure it works on your system then we can close the issue as there is no problem for me now either as the project ultimately compiled successfully! I was just bringing to your attention that this was what I had to do to get it to work.

closed 2021-01-22 by yash-fn

vintagepc, 2021-01-22:
Yes, we've not encountered this before (and the git CI system builds every commit) so there must be something strange about this particular instance.  If you're using gcc, it should just be `gcc -dumpmachine` in the command line

yysh12, 2021-01-23:
x86_64-apple-darwin20

Well that’s weird.

yysh12, 2021-01-23:
x86_64-apple-darwin20

Well that’s weird. And right now just playing around, but eventually going to attempt to use this to test an new stepper ISR I’ve been working on.

vintagepc, 2021-01-23:
Hmm... strange indeed. the only thing I can think of is the simavr module might be defaulting to a different compiler if not explicitly specified with, e.g. `cmake -DCMAKE_C_COMPILER=....`

Hope your project works out - that's exactly the reason I made this! No more guessing what might be going on in the code when something doesn't work :)

yysh12, 2021-01-23:
Yeah. Possibly something isn’t vanilla on my system. I’ve been installing a lot of random crap including all sorts of mcu toolkits and system extensions. But, as they say, if it ain’t broke.
And thanks! Yeah this system you’ve created is pretty damn amazing. I’ll be sure the spread the word! <3

yash-fn, 2021-01-23:
Yeah. Possibly something isn’t vanilla on my system. I’ve been installing a lot of random crap including all sorts of mcu toolkits and system extensions. But, as they say, if it ain’t broke.
And thanks! Yeah this system you’ve created is pretty damn amazing. I’ll be sure the spread the word! <3