vintagepc/MK404 #317: [BUG] --test flag should hide the UI in more cases

wavexx, 2021-06-24:
**Describe the bug**
I had a few scripting scenarios where I just needed to communicate with the printer by serial, or to debug it. For this reason I didn't want any UI, so I tried to use "--test -s", but it still shows the LCD emulation.

Likewise, a similar scenario is "--test --gdb", where I might be scripting gdb and thus have no need for the UI as well.
Looks like the only scenario where --test hides the UI at the moment is "--test --script".

**To Reproduce**
Start with "--test -s" or "--test --gdb".

**Expected behavior**
No extra windows being shown in these scenarios.
I almost think --test should refuse to operate if not given a compatible option (--script, -s, --gdb, or something that gives the user a way to control the board without the lcd).

vintagepc, 2021-06-24:
The "no graphics" part of that is probably wrong now as I believe in some cases we use --test with graphics for the automated testing components that take screenshots to verify the GL behaviour. 

I think what you're looking for that is more suitable to your use case is also using `-g none` as an argument. That should suppress all graphics.

wavexx, 2021-06-24:
Oh, well then, works good enough for me using --test in combination with -g none :)
Maybe it just needs a better description, or yeah removing it completely, so that graphics control is explicitly handled via -g.

wavexx, 2021-06-24:
On the other side, is there an explicit way to request the "2D ui only" via -g? I know it's default, but I think it should be listed as an explicit choice.

vintagepc, 2021-06-24:
Not currently but it should be possible to add one and just have it no-op

commit 2bdd92c4 (2021-06-27): Update help text to remove outdated info

Fixes #317
closed 2021-06-27 by vintagepc