vintagepc/MK404 #323: [BUG] Crash on SD card handling when rapid-firing 'c'

wavexx, 2021-08-13:
**Describe the bug**
MK404 crashes when holding 'c' (mount/unmount SD card)

A clear and concise description of what the bug is. Please make sure you've searched for existing bug reports before creating a new one.

**To Reproduce**
- Start MK404 with a MK3/S FW.
- Press and hold 'c' to mount/unmount the SD card quickly (I was trying to reproduce a FW issue.. believe me ;))
- Eventually MK404 itself will crash:

```
#0  0x000055a3b50c6486 in SDCard::OnSPIIn(avr_irq_t*, unsigned int) ()
[Current thread is 1 (Thread 0x7f1cb66fc700 (LWP 171312))]
(gdb) where
#0  0x000055a3b50c6486 in SDCard::OnSPIIn(avr_irq_t*, unsigned int) ()
#1  0x000055a3b50e95f8 in SPIPeripheral::OnPostInit(avr_t*, unsigned int)::{lambda(avr_irq_t*, unsigned int, void*)#1}::_FUN(avr_irq_t*, unsigned int, void*) [clone .lto_priv.0] ()
#2  0x000055a3b5115574 in avr_raise_irq_float ()
#3  0x000055a3b51163e7 in avr_spi_raise ()
#4  0x000055a3b510f9e7 in avr_cycle_timer_process ()
#5  0x000055a3b5104e3c in avr_callback_run_raw ()
#6  0x000055a3b50acbf0 in Boards::Board::RunAVR() ()
```

**Expected behavior**
No crash, I suppose :)

**Screenshots**
☠

**Desktop (please complete the following information):**
 - OS: linux

**Additional context**
No critical

vintagepc, 2021-08-13:
Good find, this is a nasty one. 

Looks like we forgot to guard against the memory-mapped data "going away" in the middle of a block read:

[image]

wavexx, 2021-08-13:
I had to laugh at the "let's never speak of this again" label :P

PR #325 (2021-08-21): Misc fixes
### Description

Minor features and fixes - a `?` hotkey and a null pointer dereference.

### Behaviour/ Breaking changes

- Prints key reference to console if user presses `?` key
-  Sim no longer crashes if holding C and trhe card unmounts while mid-read.

### Linked issues:

 - Fixes #323 
 - Fixes #324
commit a3a61b11 (2021-08-21): Misc fixes (#325)

* Implement #324

* Fix SD card state after unmount (Fixes #323)
closed 2021-08-21 by vintagepc