vintagepc/MK404 #345: [FEATURE] Commandline option for soft PWM

GilesBathgate, 2021-12-03:
**Is your feature request related to a problem? Please describe.**
When testing 3.10.x firmware on MK2 simulated hardware, I'd like the option to enable software PWM.

**Describe the solution you'd like**
 `./MK404 Prusa_MK2_mR13 --soft-pwm=enable -f ...`

If not specified, and for other printer types, this will default to how its set to currently. 

**Describe alternatives you've considered**
Maybe a new printer type specifically for this firmware? `Prusa_MK2_mR13_patched`? 

**Additional context**
This fixes the selftest bed check when running 3.10.x firmware on the MK2 sim. I can probably do the PR, if the idea makes sense.

vintagepc, 2021-12-03:
hmm, this is a more "generic" style feature so a command line flag is probably best rather than splitting off more board configurations that are 99% the same. 

My approach would probably be to add it as an item in `Config::` with a ternary state - not set, on, off and then check and use that accordingly during `Board::SetupHardware()`. 

If you have some time and would like to take a crack at a PR, please feel free! :)

vintagepc, 2021-12-04:
Closed by #346

closed 2021-12-04 by vintagepc