vintagepc/MK404 #352: [BUG] Won't start on MacOSX?

TheEvilDev, 2022-01-15:
**Describe the bug**
I'm not sure if this is a bug, or more just because I am a noob when it comes to C++, but I followed the brew instructions for MacOSX platform support, and recompiled it from source with cmake and make install. No issues with any of this, but when I run the program, I get the following output. I expect I'm missing some dependencies in the right places.

Here's the log output when I run the `MK404` cli:

```
~/w/e/MK404 ❯❯❯ ./MK404                                                                                                                                                                                      ⏎ master ✱ ◼
***************************************
*                        dev-modified *
* This is a DEV version. Features may *
* behave unexpectedly, or not at all. *
***************************************
No bootloader specified, using default: stk500boot_v2_mega2560.hex
ScriptHost: NOTE: Duplicate context name (Thermistor) with different pointer. Incrementing ID...
ScriptHost: NOTE: Duplicate context name (Thermistor) with different pointer. Incrementing ID...
ScriptHost: NOTE: Duplicate context name (Thermistor) with different pointer. Incrementing ID...
ScriptHost: NOTE: Duplicate context name (Fan) with different pointer. Incrementing ID...
Creating pinspec for atmega2560
ERROR: Could not open flash file. Flash contents will NOT persist.
ERROR: Could not open SPI flash file. Flash contents were NOT restored
ERROR: Could not open flash file. Flash contents were NOT restored
NOTE: No firmware load requested, Executing purely from flash memory: Einsy
read_ihex_chunks: stk500boot_v2_mega2560.hex, unsupported check type 03
Loaded 4202 bytes from HEX file: stk500boot_v2_mega2560.hex
Initialized VCD file Prusa_MK3S_VCD.vcd
uart_pty_init bridge on port *** /dev/ttys006 ***
/tmp/simavr-uart2 now points to: /dev/ttys006
note: export SIMAVR_UART_XTERM=1 and install picocom to get a terminal
uart_pty_init bridge on port *** /dev/ttys007 ***
uart_pty_init bridge on port *** /dev/ttys008 ***
SD file Prusa_MK3S_SDcard.bin does not exist. Will not create it.
SD card image (Prusa_MK3S_SDcard.bin) could not be mounted (error -1 ).
Init on ADC 0 start temp: 25
Init on ADC 2 start temp: 25
Init on ADC 3 start temp: 25
Init on ADC 6 start temp: 25
ScriptHost: NOTE: Duplicate context name (VSrc) with different pointer. Incrementing ID...
MK3S - adding IR sensor.
freeglut (./MK404): failed to open display ''
```

Seems like it's almost there, and my issue might just be the last line, and something isn't quite right with my freeglut setup, but I just don't have much to go on, so am hoping for some guidance.

**Desktop (please complete the following information):**
 - OS: MacOS Catalina
 - Version: 10.15.7

**Additional context**
Add any other context about the problem here - printer model, what firmware file you are running, etc.

vintagepc, 2022-01-15:
This sounds like it cannot find a display server to connect to. 

If memory serves, you need to manually start an instance of XQuartz before trying to launch MK404. If it still gives you the same error, you may need to explicitly tell it the display name to use, e.g. if the XQuartz output says something about Display :0 then the syntax is `DISPLAY=:0 ./MK404 ...`

Let me know if that fixes it, if so I'll update the documentation.

TheEvilDev, 2022-01-15:
Actually, thanks for the quick response, turns out all I needed to do was reboot, it's working now, thanks for the great work on this project!

closed 2022-01-15 by TheEvilDev

vintagepc, 2022-01-15:
Ah, fantastic! I'll add a note to the wiki.